Protocentral/ADS1292rShield_Breakout #3: bad results

ccarmielf, 2017-03-28:
i followed the steps laid out and im not getting any real results. It looks more like a beat (high and low frequency signal) then a pulse. any suggestion to what im doing wrong?

VenkateshBhat, 2017-03-29:
Hello ccarmielf,

1.  You did mention that you were facing some issues with the libraries, where did you get those instructions?
2.Recently we did update on gui and arduino sketch , could you please clone and try with the latest version
3.Please attach the obtained result.

ccarmielf, 2017-04-21:
Hi, So i got the right libraries added and the code uploads fine. Just my results are weird. 

[log.xlsx](https://github.com/Protocentral/ADS1292rShield_Breakout/files/948152/log.xlsx)

ccarmielf, 2017-04-21:
So here are the results with different lead placement.
[log.xlsx](https://github.com/Protocentral/ADS1292rShield_Breakout/files/948174/log.xlsx)

VenkateshBhat, 2017-04-24:
Hello,
The graph seems to be extremely erroneous, 

- are the electrodes placements are proper? 
- are you using shield or breakout?

preksha02, 2019-04-12:
hello VenkateshBhat 
can you help in finding the exact library for this.

yasaswinir, 2019-04-12:
Hello Preksha,
You can find the arduino library in the software/ folder of our github repo

preksha02, 2019-05-14:
https://photos.app.goo.gl/3jHZedEEafnVNs7H9
this is the link of my output. i dont get respiration rate, neither on arduino ide nor on processing

closed 2020-12-31 by protocentralashwin