Protocentral/ADS1292rShield_Breakout #4: Never get DRDY

philgrim2, 2017-07-19:
Using the breakout board with an Uno R3.
Put in some Serial.println settings to try to discover why I don't get results.
DRDY never goes low.
I have two boards, same results with both.

VenkateshBhat, 2017-07-19:
Could you please attach the connection setup?

gcmike, 2017-09-29:
Same issue here. Any updates?

VenkateshBhat, 2017-09-29:
@gcmike 
Sorry for the inconvenience caused,
Please provide us the additional details/share the screenshots.

VenkateshBhat, 2017-09-30:
@gcmike thanks for sharing,
Couple of things:
- If your msp430xxx is 3.3V tolerable change the jumper setting on the breakout to 3.3V. By default it will be 5V.
- No additional pulldown is required since there is a on board level converter.
Ensure START and PWDN/RESET are pulled high.
[image]

closed 2020-12-31 by protocentralashwin