Protocentral/ADS1292rShield_Breakout #7: Latest Version not showing any data on Windows 10

twhiting87, 2018-01-16:
Hi, Using the latest versions of Arduino and Processing code from Github (both compile and run correctly), the GUI screen is loading but, after choosing the correct COM port (4), there is no  data being passed through - see screen shot. I've checked the serial port using RealTerm and good data is coming through so the issue is somewhere in the Processing code. Any ideas?

[image]

kalineBr, 2018-01-16:
Maybe the serial.begin isn't synchronized: look at devices an printers->drivers->arduino->properties->serial.

shreythakker, 2018-06-08:
Facing the Same problem. was your issue resolved?

preksha02, 2019-05-14:
i am not able to compile processing code. Can you please send your processing code. @twhiting87

joicetm, 2019-05-14:
Hi,
Did you follow all procedure what we given in our official website(https://github.com/Protocentral/ADS1292rShield_Breakout),if you done those procedure please share the screenshot where you are getting error even though please share your order number.

preksha02, 2019-05-14:
Thankyou for replying!
https://photos.app.goo.gl/3jHZedEEafnVNs7H9
this is my output wave form.
i get the heart rate value, but not respiration rate value.
even though in arduino ide,i only get heart rate. there also a issue , the code i am using doesnt show values. here i am attaching my arduino ide output.
https://photos.google.com/photo/AF1QipNO8xMagkwqkP0eVKDIAztxl8fGNsTPNn8_oofC
my arduino program->
//////////////////////////////////////////////////////////////////////////////////////////
//
//   Arduino Library for ADS1292R Shield/Breakout
//
//   Copyright (c) 2017 ProtoCentral
//   Heartrate and respiration computation based on original code from Texas Instruments
//
//   This software is licensed under the MIT License(http://opensource.org/licenses/MIT).
//
//   THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT
//   NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.
//   IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
//   WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE
//   SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
//
//   Requires g4p_control graphing library for processing.  Built on V4.1
//   Downloaded from Processing IDE Sketch->Import Library->Add Library->G4P Install

// If you have bought the breakout the connection with the Arduino board is as follows:
//
//|ads1292r pin label| Arduino Connection   |Pin Function      |
//|----------------- |:--------------------:|-----------------:|
//| VDD              | +5V                  |  Supply voltage  |
//| PWDN/RESET       | D4                   |  Reset           |
//| START            | D5                   |  Start Input     |
//| DRDY             | D6                   |  Data Ready Outpt|
//| CS               | D7                   |  Chip Select     |
//| MOSI             | D11                  |  Slave In        |
//| MISO             | D12                  |  Slave Out       |
//| SCK              | D13                  |  Serial Clock    |
//| GND              | Gnd                  |  Gnd             |
//
/////////////////////////////////////////////////////////////////////////////////////////

#include <ads1292r.h>
#include <SPI.h>

ads1292r ADS1292;   // define class

//Packet format
#define  CES_CMDIF_PKT_START_1      0x0A
#define  CES_CMDIF_PKT_START_2      0xFA
#define  CES_CMDIF_TYPE_DATA        0x02
#define  CES_CMDIF_PKT_STOP_1       0x00
#define  CES_CMDIF_PKT_STOP_2       0x0B

#define TEMPERATURE 0
#define FILTERORDER         161
/* DC Removal Numerator Coeff*/
#define NRCOEFF (0.992)
#define WAVE_SIZE  1

//******* ecg filter *********
#define MAX_PEAK_TO_SEARCH         5
#define MAXIMA_SEARCH_WINDOW      25
#define MINIMUM_SKIP_WINDOW       30
#define SAMPLING_RATE             125
#define TWO_SEC_SAMPLES           2 * SAMPLING_RATE
#define QRS_THRESHOLD_FRACTION    0.4
#define TRUE 1
#define FALSE 0

volatile uint8_t  SPI_Dummy_Buff[30];
[... 453 lines cut ...]
    {
      /* Reset heart rate computation sate variable in case of no peak found in 3 seconds */
      sample_count = 0 ;
      s_array_index = 0 ;
      m_array_index = 0 ;
      maxima_sum = 0 ;
      sample_index[0] = 0 ;
      sample_index[1] = 0 ;
      sample_index[2] = 0 ;
      sample_index[3] = 0 ;
      Start_Sample_Count_Flag = 0;
      peak_detected = FALSE ;
      sample_sum = 0;
      first_peak_detect = FALSE;
      nopeak = 0;
      QRS_Heart_Rate = 0;
    }
  }

  global_HeartRate = (uint8_t)QRS_Heart_Rate;
  //Serial.println(global_HeartRate);

}

kindly help!

joicetm, 2019-05-14:
please share your order number so that we can contact you directly to solve your issue

preksha02, 2019-05-14:
i am not getting it. what is order number? @joicetm

Mo7amed-Aazeem, 2019-10-07:
joicetm..
I have a problem with getting output from this circuit, after i following all of the mentioned steps,
can you help me please...
as fast as possible please

Email: Mo7amed99935@gmail.com 
Whatsapp: +249100100409

Mo7amed-Aazeem, 2019-10-08:
> please share your order number so that we can contact you directly to solve your issue

help me please...

MohammedHM, 2019-10-09:
Hello Aazeem,

Our ADS1292R Arduino library calculates heart rate, ECG waveform and Respiration waveform. We did not add respiration algorithm to calculate respiration value due to memory sufficiency while compiling in Arduino UNO. You can get the respiration algorithm from our [HealthyPi Arduino library ](https://github.com/Protocentral/protocentral-healthypi-v3/blob/master/firmware/healthypi3_streaming/healthypi3_streaming.ino) and you can add this algorithm and try with some boards with more memory like Arduino mega.

MohammedHM, 2019-10-09:
Hello Aazeem,

Please share the screenshots of your GUI and the error your facing to get output. Surely we can help you in figuring it out.

Mo7amed-Aazeem, 2019-10-09:
Thank you for replying… I'm trying to implement a system to measure ECG signal + respiration + heartrate..so I chose to work with your ads1292r, I follow all of the steps that mentioned in the site,  but I have no output at all,  some times the output come-out as a very noisy signal (I will send you a screenshot for it) and sometimes there is no output but the noisy signal appears with body movement and disappear again. But usually there is no response…Actually I was read about the module and the project before getting it,  so I decide to use the arduino due,  in stead of uno because of the better specifications of it(memory,  speed and so on) Sent from my Huawei Mobile-------- Original Message --------Subject: Re: [Protocentral/ADS1292rShield_Breakout] Latest Version not showing any data on Windows 10 (#7)From: Mohammed Ibrahim To: Protocentral/ADS1292rShield_Breakout CC: Mo7amed-Aazeem ,Comment Hello Aazeem,
Our ADS1292R Arduino library calculates heart rate, ECG waveform and Respiration waveform. We did not add respiration algorithm to calculate respiration value due to memory sufficiency while compiling in Arduino UNO. You can get the respiration algorithm from our HealthyPi Arduino library  and you can add this algorithm and try with some boards with more memory like Arduino mega.

Mo7amed-Aazeem, 2019-10-09:
Sent from my Huawei Mobile-------- Original Message --------Subject: Re: [Protocentral/ADS1292rShield_Breakout] Latest Version not showing any data on Windows 10 (#7)From: Mohammed Ibrahim To: Protocentral/ADS1292rShield_Breakout CC: Mo7amed-Aazeem ,Comment Hello Aazeem,
Our ADS1292R Arduino library calculates heart rate, ECG waveform and Respiration waveform. We did not add respiration algorithm to calculate respiration value due to memory sufficiency while compiling in Arduino UNO. You can get the respiration algorithm from our HealthyPi Arduino library  and you can add this algorithm and try with some boards with more memory like Arduino mega.

closed 2020-12-31 by protocentralashwin