Protocentral/ADS1292rShield_Breakout #9: global_respirationRate is null

kaline, 2018-03-21:
Inside the example code is not possible measure the respiration, the monitor serial always print '0':
Somebody has any ideas about the problem?

The piece of code:

  Serial.print("heart_rate->");
  Serial.println(global_HeartRate);
  Serial.print("respiration_rate-> ");
  Serial.println(global_RespirationRate);

pzink, 2018-11-23:
The most recent example code doesn't include any output of respiration rate. Is there older or better example code that includes that?

yasaswinir, 2018-11-24:
Hello @pzink @kaline , 
Sorry for the inconvenience. We are trying to fix the issues in both GUI and the code. Meanwhile if you want to check the respiration rate , you can verify in the log file created after you click record in the Processing GUI. The respiration rate is saved into the log file.

Thank you for bringing the issue forward!

preksha02, 2019-05-14:
@yasaswinir 
whether the issue resolve? because still i am facing the same problem. I am only able to view ECG not the respiration in arduino IDE.

yingshaoxo, 2020-10-04:
whether the issue resolve? because still i am facing the same problem. I am only able to view ECG not the respiration in arduino IDE.