Protocentral/ADS1292rShield_Breakout #11: ECG lead error!!! ensure the leads are properly connected

davidm-public, 2018-11-02:
Hi, uploaded the `text_only' arduino file, got the following:

```
Initiliziation is done
ECG lead error!!! ensure the leads are properly connected
ECG lead error!!! ensure the leads are properly connected
ECG lead error!!! ensure the leads are properly connected
etc.
```

thing is that I've placed thousands of ecg electrodes in my time, and I can honestly say that these leads are indeed connected !

yasaswinir, 2018-11-05:
Hello David sir! Glad you could figure out the issue! Thank you so much for your acknowledgment! Please do let us know if we could be of any further service to you!

davidm-public, 2018-11-18:
Hi there, looks like your website is down.
I'd like to order a few more of those boards.

David

closed 2018-11-18 by davidm-public