Protocentral/ADS1292rShield_Breakout #14: Error while Compiling

shabir67, 2020-02-23:
Just get the package recently any body knows hot to deal with this error message ??
Arduino: 1.8.5 (Windows 10), Board: "Arduino Uno"

> D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\SHABIR~1\AppData\Local\Temp\arduino_build_774088 -warnings=none -build-cache C:\Users\SHABIR~1\AppData\Local\Temp\arduino_cache_153723 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=D:\Arduino\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=D:\Arduino\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\Shabir As\Desktop\TA\Proto Central\Protocentral_ADS1292R\examples\ECG-Respiration-Plot\ECG-Respiration-Plot.ino
> D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\SHABIR~1\AppData\Local\Temp\arduino_build_774088 -warnings=none -build-cache C:\Users\SHABIR~1\AppData\Local\Temp\arduino_cache_153723 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=D:\Arduino\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=D:\Arduino\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=D:\Arduino\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\Shabir As\Desktop\TA\Proto Central\Protocentral_ADS1292R\examples\ECG-Respiration-Plot\ECG-Respiration-Plot.ino
> Using board 'uno' from platform in folder: D:\Arduino\portable\packages\arduino\hardware\avr\1.8.2
> Using core 'arduino' from platform in folder: D:\Arduino\portable\packages\arduino\hardware\avr\1.8.2
> Detecting libraries used...
> "D:\Arduino\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-ID:\Arduino\portable\packages\arduino\hardware\avr\1.8.2\cores\arduino" "-ID:\Arduino\portable\packages\arduino\hardware\avr\1.8.2\variants\standard" "C:\Users\SHABIR~1\AppData\Local\Temp\arduino_build_774088\sketch\ECG-Respiration-Plot.ino.cpp" -o "nul"
> "D:\Arduino\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-ID:\Arduino\portable\packages\arduino\hardware\avr\1.8.2\cores\arduino" "-ID:\Arduino\portable\packages\arduino\hardware\avr\1.8.2\variants\standard" "C:\Users\SHABIR~1\AppData\Local\Temp\arduino_build_774088\sketch\ECG-Respiration-Plot.ino.cpp" -o "C:\Users\SHABIR~1\AppData\Local\Temp\arduino_build_774088\preproc\ctags_target_for_gcc_minus_e.cpp"
> C:\Users\Shabir As\Desktop\TA\Proto Central\Protocentral_ADS1292R\examples\ECG-Respiration-Plot\ECG-Respiration-Plot.ino:20:10: fatal error: ads1292r.h: No such file or directory
> 
>  #include <ads1292r.h>
> 
>           ^~~~~~~~~~~~
> 
> compilation terminated.
> 
> exit status 1
> Error compiling for board Arduino Uno.

MarijnStam, 2020-02-24:
You have to include the library in your Arduino enviroment.

Sketch -> Include Library -> Add .ZIP library 
and then select the libraries which you can download [here](https://github.com/Protocentral/ADS1292rShield_Breakout/releases/download/v1.0/Protocentral_ADS1292R-v1.0-arduino.zip)

shabir67, 2020-03-12:
Thank you very much.One more question, is it possible to send this data package to and MySQL DB with esp8266 module ??.