Protocentral/ADS1292rShield_Breakout #17: Read the respiration data to Arduino serial console.

desertwire, 2020-04-28:
Is there any way to read the value of the respiration and heart rate without using the Protocentral GUI. Printing DataPacketHeader to the console just results in corrupted ASCII.

```C
DataPacketHeader[9] = s32DaqVals[0];            // 4 bytes Respiration data
DataPacketHeader[10] = s32DaqVals[0]>>8;
DataPacketHeader[11] = s32DaqVals[0]>>16;
DataPacketHeader[12] = s32DaqVals[0]>>24; 

    for(i=0; i<15; i++) 
    {
      Serial.write(i);
      Serial.write(DataPacketHeader[i]);     // transmit the data over USB
    } 
```

Here is the output below of the Arduino serial monitor, is there any way to make sense of the data or how it is converted to a readable integer on the GUI?

```
⸮ < 0 p s | ` r I @ @ 9   	⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ < 2 $ & O ` q ` 8 0 ! > ⸮ ⸮ ⸮ ⸮ ⸮  ! 8 ! | H a @  9 $  ! ⸮ 8 B x  ⸮ ⸮ ⸮ ⸮  y a ` p 	f ⸮ ⸮ ⸮ ⸮ ⸮ 1 b N p ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ " D ` ` ` p  
 ! ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ 0 ⸮ ' # 3 @ O ` D B @ @ ! ⸮ ⸮ ⸮ ⸮ ⸮ 0 	a F ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ " x p $ 0 ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ? q ` c x L b 8 9 $ 	⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ 0 " ' & ⸮ & 8 # s ` a f H @ p | c N ~ "  ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ @ D O D " ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ ⸮ $  F @ p 0 2 ⸮ ⸮ 
```

desertwire, 2020-04-28:
[Link](https://arduino.stackexchange.com/questions/75216/read-the-respiration-data-to-arduino-serial-console) to discussion.

protocentralashwin, 2020-12-31:
From the previous comment by @ryanjonesxyz, this will be in the format as required by [ProtoCentral OpenView](https://github.com/Protocentral/protocentral_openview)

closed 2020-12-31 by protocentralashwin